fenrus75/powertop #32: Release 2.11 availability on https://01.org/powertop/downloads

ffontaine, 2019-09-28:
Do you plan on adding release 2.11 to https://01.org/powertop/downloads? I can't built the tarball retrieved from https://github.com/fenrus75/powertop/releases/tag/v2.11 because it misses the configure script and autoreconf does not work because it also misses the version-long file.

lkraav, 2019-09-29:
Also, is there a changelog anywhere? I don't see any NEWS or CHANGELOG files in the tree, nor anything in GH release tag description.

nswamina, 2019-10-02:
We have autogen.sh script and configure script that can be used to install. configure script is available in the repo. The tarball has been updated on the 01.org/powertop downloads page too.

gkammela, 2019-10-02:
@ffontaine, please run ./autogen.sh to get the configure files generated. I had tried building from the tarball and it works perfectly fine.

gkammela, 2019-10-02:
@lkraav we have updated the changelog here https://01.org/powertop/downloads/powertop-v2.11-0. Please check.

ffontaine, 2019-10-02:
Thanks for making the tarball available and pointing out that autogen.sh should be run, I'll close this issue.

closed 2019-10-02 by ffontaine