fenrus75/powertop #34: `-C` and `-r` are not working

dac73, 2019-10-20:
```
powertop -C --debug
modprobe cpufreq_stats failedLoaded 67 prior measurements
RAPL device for cpu 0
RAPL device for cpu 0
Devfreq not enabled
glob returned GLOB_ABORTED
Preparing to take measurements
To show power estimates do 344 measurement(s) connected to battery only
Segmentation fault
```
```
powertop -r --debug
modprobe cpufreq_stats failedLoaded 67 prior measurements
RAPL device for cpu 0
RAPL device for cpu 0
Devfreq not enabled
glob returned GLOB_ABORTED
Preparing to take measurements
To show power estimates do 344 measurement(s) connected to battery only
Segmentation fault
```

jubalh, 2019-10-21:
Works fine with `2.11-1-g7ef7f79` on my machine with openSUSE.

dac73, 2019-10-21:
Oh, let me add more context :)
It works for me also on Asus with intel. It fails on ThinkPad with AMD (Rayzen 3700u).
Fedora 30@5.3.6-200.fc30.x86_64

dr-br, 2019-11-12:
Does not work:
System: AMD Athlon 200GE, Ubuntu 19.10, 5.0.0-32-generic x86_64
PowerTOP: 7ef7f7974e781590edca22ff4451b9fd7eafdf3d (v2.11)

strace: 
openat(AT_FDCWD, "/proc/timer_stats", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)

dac73, 2020-05-02:
More info:
```
May 02 10:50:30 kraken systemd-coredump[5351]: Process 5345 (powertop) of user 0 dumped core.
                                               
                                               Stack trace of thread 5345:
                                               #0  0x00005568742b54e4 _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm.isra.0 (powertop + 0x254e4)
                                               #1  0x00005568742ba3b7 _Z26report_display_cpu_cstatesv (powertop + 0x2a3b7)
                                               #2  0x00005568742ac99d _Z15one_measurementiiPc (powertop + 0x1c99d)
                                               #3  0x00005568742aca7f _Z11make_reportiPciiS_ (powertop + 0x1ca7f)
                                               #4  0x00005568742a14ad main (powertop + 0x114ad)
                                               #5  0x00007f15a5833042 __libc_start_main (libc.so.6 + 0x27042)
                                               #6  0x00005568742a258e _start (powertop + 0x1258e)
```

eckelj, 2022-08-29:
this happens also on 
Packages:
	0: Intel Xeon Platinum 8259CL
Cores:
	0: 2 processors (0-1), Intel Sky Lake
btw. I am running ubuntu