fenrus75/powertop #36: make - no such a file or directory.

nnmlss, 2020-01-05:
I'm trying to update my version following the commands listed into the md file but when I try to run `./make` I receive an output - no such a file or directory. What do I do wrong? I have installed all the sample packages for Ubuntu, I'm using elementary OS Hera, based on Ubuntu 18.04

Phil171, 2020-03-28:
Same to me, tried `make` only, doesn't works. Only previous version from Ubuntu website works.
I can run `./autogen.sh` and `./configure` successfully, after, I cannot continue.
Xubuntu 18.04
Thanks, Phil from France

gkammela, 2020-04-24:
@nnmlss and @Phil171, do you still see the issue with make command?

nnmlss, 2020-04-24:
yes, I have tried a few day ago and received the same results

Phil171, 2020-04-25:
Sorry, I'm out of the game right now, problems with my brand new Dell XPS 15 which consumes a lot of randomness in Linux and reinstall with Xubuntu 20.04 to remove problems from V18.04 without success, alas ...
The other 'little problem' is that I am not very familiar with Linux, I am more of an average user, user of the kind 'copy / paste' an instruction in a tutorial, sometimes I can not understand the subtleties of some command lines and I block, I'm stuck. And there is also the (big) problem of Github which is generally addressed to 'mustaches', people who know who it is ...
If the instructions are detailed enough and explained, it is best to follow the installation of a program.

But I'm learning, you have to be basic. On the other hand, I had the same problems as nnmiss, the same errors during my first post. Sorry for my poor or bad english...

thac0, 2020-05-13:
Acknowledged the issue about autoconf's "too many loops" error. That will be investigated separately. If "no such file or directory" comes back, let us know.

closed 2020-05-13 by thac0