fenrus75/powertop #37: Powertop does not remember turntables at reboot

WolfpackN64, 2020-01-19:
I run Powertop with autotune on my Thinkpad T440, running Fedora 31 XFCE. As I basically want all turntables to be set on "Good". However, my USB mouse connected to my laptop freezes sporadically when it's turntable is set to "Good", so the USB autosuspend for my USB mouse is the is the only turntable I turn to "Bad". However, the turntable changes don't survive reboot, and no forum or post anywhere has a good, universal solution to this problem.

It would be far easier for end-users if Powertop could permanently write the turntable chances to the system so that your turntables survive rebooting, and it would make enabling and disabling the turntables for all kind of settings much easier.

Making complicated custom scripts for each mouse I'm going to use on my laptop, which might not work on different distros really is a bit much.

pflanzenandi, 2020-05-13:
I suppose that many users of Powertop wish for exactly these features:

1. tunables set on startup
2. exclude unwanted tunables like usb devices

thomaseizinger, 2020-06-21:
The following works well for me as a workaround:

```
[Unit]
Description=Override tunables
After=powertop.service

[Service]
Type=oneshot
ExecStart=/usr/bin/bash -c 'echo on > /sys/bus/usb/devices/3-2.3/power/control'
ExecStart=/usr/bin/bash -c 'echo on > /sys/bus/usb/devices/3-2.4/power/control'

[Install]
WantedBy=multi-user.target
```

Several things to be noted here which weren't obvious to me as a not-so-experienced systemd user :)

1. Multiple `ExecStart` only work with type `oneshot`
2. `ExecStart` doesn't use a shell, i.e. you can directly redirect from there to the file, need to invoke `bash` for that.

hellow554, 2021-02-03:
hey @fenrus75 
Is there a plan on implementing this without workarounds?