fenrus75/powertop #39: README: Clarify ondemand requirement

pacho2, 2020-02-23:
I was reviewing README to know the kernel options I need to enable for running powertop, but I was a bit surprised about ondemand governor requirement. It's a long time since I disabled it in favor of pstate driver. Is README updated? Is ondemand really required even on recent intel processors?

Thanks for the info

gkammela, 2020-04-24:
Hi @pacho2, @thac0 is doing a major rewrite to README. I hope that clarifies the requirements needed.

thac0, 2020-05-04:
The short answer is "no", @pacho2 . The `ondemand` governor is not required.

While my initial patch volley will not audit the required kernel options, a subsequent one will.

pacho2, 2021-09-08:
Thanks

Do you plan to get it handled for the next version? 

Best regards