fenrus75/powertop #41: I had shutdown my computer while running calibration (my pc is fine but I just wanna know)

Rowisi, 2020-04-03:
I ran `powertop --calibrate` command and the screen suddenly gone black, I though my computer had crashed so I forced shutdown by holding power button. (my fault I didn't read the wiki well)

My computer is fine now but I just wanna know if there are some side effects. Although I know that changes to /sys directory aren't permanent.

I have rerun calibration again until it became successful to make sure everything is okay.

**My Suggestion:** Can you add a confirmation request by `y` yes or `n` no with a warning that a the screen back light will turn off during measurements before running the command. It will be safer for users.

Cheers.

Rowisi, 2020-04-05:
The issue was a bit exaggerated. No problems so far. Closing.

closed 2020-04-05 by Rowisi

D-EVE-R, 2022-09-18:
> I ran `powertop --calibrate` command and the screen suddenly gone black, I though my computer had crashed so I forced shutdown by holding power button. (my fault I didn't read the wiki well)

The same thing happened to me and when I did back I noticed that the brightness keys F11 and F12 stopped working.