fenrus75/powertop #60: powertop exits at startup with read_msr failure

hughsie, 2020-06-16:
I'm using Powertop in Fedora 32 on a Lenovo P720 machine. It has 2 physical Intel Xeon Gold 6130 CPUs @ 2.1GHz x 64.

    $ sudo powertop
    the port is sda
        the port is sdb
            read_msr cpu0 0x630 : Input/output error
    $ echo $?
    254

thac0, 2020-06-16:
Thanks for letting us know, @hughsie . Yuck.

For sanity, can you do a quick `cat /proc/cpuinfo` and, for one of the CPUs, provide the `cpu family:`, `model:`, `model name:`, and `stepping:` fields?

(DMing them to me is also an option).

hughsie, 2020-06-16:
Here you go:

    processor    : 0
    vendor_id    : GenuineIntel
    cpu family    : 6
    model        : 85
    model name    : Intel(R) Xeon(R) Gold 6130 CPU @ 2.10GHz
    stepping    : 4
    microcode    : 0x2000065
    cpu MHz        : 1000.062
    cache size    : 22528 KB
    physical id    : 0
    siblings    : 32
    core id        : 0
    cpu cores    : 16
    apicid        : 0
    initial apicid    : 0
    fpu        : yes
    fpu_exception    : yes
    cpuid level    : 22
    wp        : yes
    flags        : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l3 cdp_l3 invpcid_single pti intel_ppin ssbd mba ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm mpx rdt_a avx512f avx512dq rdseed adx smap clflushopt clwb intel_pt avx512cd avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts hwp hwp_act_window hwp_epp hwp_pkg_req pku ospke md_clear flush_l1d
    vmx flags    : vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple shadow_vmcs pml ept_mode_based_exec tsc_scaling
    bugs        : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs taa itlb_multihit
    bogomips    : 4200.00
    clflush size    : 64
    cache_alignment    : 64
    address sizes    : 46 bits physical, 48 bits virtual
    power management:

thac0, 2020-06-16:
@hughsie - As I thought. I ran into the same issue not too long ago, so I did #53 . Basically, prior to the v2.13 release, that model was incorrectly defined, so it was attempting to read a (non-existent on that model) `MSR_PKG_C8`. That patch dodges the more insidious bug by not attempting to read it (and a couple others) in the first place.

thac0, 2020-06-16:
The "more insidious bug" is that PowerTOP doesn't have any smarts when it encounters a non-existent MSR (and happily reads it).

ElCoyote27, 2021-05-30:
Seeing the same issue on RHEL8.4 and Xeon Platinum 8269CY's..

```
 the port is sda
                 the port is sdb
                                 the port is sdc
                                                read_msr cpu0 0x630 : Input/output error
```

ElCoyote27, 2021-05-30:
```# lscpu 
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              104
On-line CPU(s) list: 0-103
Thread(s) per core:  2
Core(s) per socket:  26
Socket(s):           2
NUMA node(s):        2
Vendor ID:           GenuineIntel
BIOS Vendor ID:      Intel
CPU family:          6
Model:               85
Model name:          Intel(R) Xeon(R) Platinum 8269CY CPU @ 2.50GHz
BIOS Model name:     Intel(R) Xeon(R) Platinum 8269CY CPU @ 2.50GHz
Stepping:            7
CPU MHz:             1200.066
CPU max MHz:         3800.0000
CPU min MHz:         1200.0000
BogoMIPS:            5000.00
Virtualization:      VT-x
L1d cache:           32K
L1i cache:           32K
L2 cache:            1024K
L3 cache:            36608K
NUMA node0 CPU(s):   0,2,4,6,8,10,12,14,16,18,20,22,24,26,28,30,32,34,36,38,40,42,44,46,48,50,52,54,56,58,60,62,64,66,68,70,72,74,76,78,80,82,84,86,88,90,92,94,96,98,100,102
NUMA node1 CPU(s):   1,3,5,7,9,11,13,15,17,19,21,23,25,27,29,31,33,35,37,39,41,43,45,47,49,51,53,55,57,59,61,63,65,67,69,71,73,75,77,79,81,83,85,87,89,91,93,95,97,99,101,103
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l3 cdp_l3 invpcid_single intel_ppin ssbd mba ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid cqm mpx rdt_a avx512f avx512dq rdseed adx smap clflushopt clwb intel_pt avx512cd avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts hwp hwp_act_window hwp_epp hwp_pkg_req pku ospke avx512_vnni md_clear flush_l1d arch_capabilities
```

kobuworks, 2021-11-15:
So powertop does not support Xon-SP, correct? thanks,
BR/Kobu