fenrus75/powertop #62: Stats for display backlight being reported as 0mW

marcelopm, 2020-06-20:
Hello

I was wondering what do I need do, so that powertop will properly read stats for my laptop's (lenovo e531) display.
My if it's relevant, my GPU is a hybrid intel HD 4000/nvidia 740m one. I've already tried with both.

Currently this is how it looks like: 19.9%;Display backlight;    0 mW

I already tried setting the kernel's option acpi_backlight=vendor, but it made no difference.

These is how my current relevant kernel's config flags looks like:
```bash
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_NO_HZ
CONFIG_NO_HZ_COMMON=y
CONFIG_NO_HZ_IDLE=y
# CONFIG_NO_HZ_FULL is not set
CONFIG_NO_HZ=y
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_HIGH_RES_TIMERS
CONFIG_HIGH_RES_TIMERS=y
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_HPET_TIMER
CONFIG_HPET_TIMER=y
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_CPU_FREQ_GOV_ONDEMAND
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_USB_SUSPEND
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_SND_AC97_POWER_SAVE
CONFIG_SND_AC97_POWER_SAVE=y
CONFIG_SND_AC97_POWER_SAVE_DEFAULT=0
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_TIMER_STATS
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_PERF_EVENTS
CONFIG_PERF_EVENTS=y
CONFIG_PERF_EVENTS_INTEL_UNCORE=y
CONFIG_PERF_EVENTS_INTEL_RAPL=m
CONFIG_PERF_EVENTS_INTEL_CSTATE=m
# CONFIG_PERF_EVENTS_AMD_POWER is not set
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_EVENT_POWER_TRACING_DEPRECATED
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_X86_MSR
CONFIG_X86_MSR=m
$ cat /lib/modules/$(uname -r)/build/.config | grep ACPI_PROCFS_POWER
# CONFIG_ACPI_PROCFS_POWER is not set
$ cat /lib/modules/$(uname -r)/build/.config | grep CONFIG_DEBUG_FS
CONFIG_DEBUG_FS=y
```
Bebug report:
```bash
____________________________________________________________________
			P o w e r T O P

____________________________________________________________________
 *  *  *   System Information   *  *  *

PowerTOP Version;v2.11-1-g7ef7f79 ran at Sat Jun 20 14:50:09 2020

Kernel Version;Linux version 5.4.0-37-generic
System Name;LENOVO68851L2ThinkPad Edge E531
CPU Information;4 Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
OS Information;Ubuntu 20.04 LTS

Target: 1 units/s;System:  196.1 wakeup/s;CPU:  3.2% usage;GPU: 0 ops/s;GFX: 5.9 wakeups/s;VFS: 0 ops/s;

____________________________________________________________________
 *  *  *   Top 10 Power Consumers   *  *  *

Usage;Events/s;Category;Description;PW Estimate
100.0%;;Device;Audio codec hwC0D0: Conexant;  500 mW  
100.0%;;Device;Audio codec hwC0D3: Intel;  500 mW  
  0.5%; 31.1;Process;[PID 3541] /usr/lib/firefox/firefox ;  130 mW  
  0.0%; 22.6;Timer;tick_sched_timer; 89.8 mW  
  0.2%; 18.3;Process;[PID 3553] /usr/lib/firefox/firefox ; 74.9 mW  
  0.3%; 13.5;Process;[PID 4017] /usr/lib/firefox/firefox -contentproc -childID 12 -isForBrowser -prefsLen 7433 -prefMapSize 227248 -parentBuildID 20; 58.3 mW  
  0.3%; 11.4;Process;[PID 3903] /usr/lib/firefox/firefox -contentproc -childID 9 -isForBrowser -prefsLen 7361 -prefMapSize 227248 -parentBuildID 202; 49.4 mW  
  0.3%;  9.4;Process;[PID 2493] cinnamon --replace ; 46.5 mW  
[... 454 lines cut ...]
Autosuspend for unknown USB device 1-1 (8087:0024)
Autosuspend for USB device BCM20702A0 [Broadcom Corp]
Autosuspend for USB device EHCI Host Controller [usb1]
Autosuspend for unknown USB device 2-1 (8087:0024)
Autosuspend for USB device xHCI Host Controller [usb3]
Runtime PM for PCI Device NVIDIA Corporation GK208M [GeForce GT 740M]
Runtime PM for PCI Device Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port

____________________________________________________________________
 *  *  *   Wake status of the devices   *  *  *

Description;Script
Wake-on-lan status for device enp5s0;echo 'enabled' > '/sys/class/net/enp5s0/device/power/wakeup';
Wake-on-lan status for device wlp4s0;echo 'enabled' > '/sys/class/net/wlp4s0/device/power/wakeup';
Wake status for USB device usb3;echo 'enabled' > '/sys/bus/usb/devices/usb3/power/wakeup';
Wake status for USB device 2-1;echo 'enabled' > '/sys/bus/usb/devices/2-1/power/wakeup';
Wake status for USB device usb1;echo 'enabled' > '/sys/bus/usb/devices/usb1/power/wakeup';
Wake status for USB device 1-1.3;echo 'enabled' > '/sys/bus/usb/devices/1-1.3/power/wakeup';
Wake status for USB device 1-1;echo 'enabled' > '/sys/bus/usb/devices/1-1/power/wakeup';
Wake status for USB device usb4;echo 'enabled' > '/sys/bus/usb/devices/usb4/power/wakeup';
Wake status for USB device usb2;echo 'enabled' > '/sys/bus/usb/devices/usb2/power/wakeup';
Wake status for USB device 3-3;echo 'enabled' > '/sys/bus/usb/devices/3-3/power/wakeup';
```

Thanks

gkammela, 2020-06-24:
Hi @marcelopm, thanks for reporting the issue. I'm currently trying to reproduce the issue and will post a fix.

fenrus75, 2020-06-24:
in part, powertop needs measurements of the screen on and screen off
situation; without that it cannot do a model fit on estimate the power....

gkammela, 2020-06-25:
@marcelopm, please follow @fenrus75's suggestion and run a few screen off and screen on tests and then check the power estimate

marcelopm, 2020-06-29:
@gkammela sorry, what do you mean by "run a few screen off and screen on tests" ?
I tried closing my laptop's screen lid, waiting a few seconds and reopening it and repeated the process a few times while running powertop, but it doesn't seem to change anything.

gkammela, 2020-07-07:
@marcelopm, please use these commands below for screen off and screen on
xset -d 0.0 dpms force off 
xset -d 0.0 dpms force on

marcelopm, 2020-07-19:
@gkammela I can make the display to go off by using:
```bash
xset -display :0 dpms force off
```
However, as I hit key on my keyboard, the display goes on automatically.

I did that a few times and, just after the display goes on, I can see that powertop stated 0%, which makes sense.
But then it just goes back to where it was before:
```bash
    0 mW     29.9%        Display backlight
```