fenrus75/powertop #64: Segmentation fault when trying to create HTML report

Mershl, 2020-07-24:
PowerTOP version 2.13
Kernel: 5.7.9 (Fedora 32)

Trying to create an HTML report with `sudo powertop --html` results in the following output:

```
modprobe cpufreq_stats failedLoaded 421 prior measurements
RAPL device for cpu 0
RAPL device for cpu 0
Devfreq not enabled
glob returned GLOB_ABORTED
Preparing to take measurements
[1]    15086 segmentation fault  sudo powertop --html
```

jubalh, 2020-07-24:
Works fine for me:

```
powertop --html
modprobe cpufreq_stats failedLoaded 0 prior measurements
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask f
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask f
Devfreq not enabled
glob returned GLOB_ABORTED
Preparing to take measurements
Taking 1 measurement(s) for a duration of 20 second(s) each.
 the port is sda
PowerTOP outputting using base filename powertop.html
```

I guess this depends on your hardware. Can you run powertop in gdb or upload a coredump?

Mershl, 2020-07-24:
I reckon it's a hardware issue. Tested it on an Lenovo Y50 without problems. My T495 is showing the following stack trace (bt full):

```
   #0  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_set_length (__n=0, this=0x4fb818)
    at /usr/src/debug/gcc-10.1.1-1.fc32.x86_64/obj-x86_64-redhat-linux/x86_64-redhat-linux/libstdc++-v3/include/bits/basic_string.h:217
No locals.
#1  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace (this=this@entry=0x4fb818, __pos=__pos@entry=0, 
    __len1=<optimized out>, __s=__s@entry=0x46474f "", __len2=__len2@entry=0)
    at /usr/src/debug/gcc-10.1.1-1.fc32.x86_64/obj-x86_64-redhat-linux/x86_64-redhat-linux/libstdc++-v3/include/bits/basic_string.tcc:470
        __old_size = <optimized out>
        __new_size = 0
#2  0x000000000041aaa6 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign (__s=0x46474f "", this=0x4fb818)
    at /usr/include/c++/10/bits/basic_string.h:901
No locals.
#3  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator= (__s=0x46474f "", this=0x4fb818)
    at /usr/include/c++/10/bits/basic_string.h:676
No locals.
#4  report_display_cpu_cstates () at cpu/cpu.cpp:564
        first_cpu = true
--Type <RET> for more, q to quit, c to continue without paging--c
        first_core = <optimized out>
        num_cpus = <optimized out>
        num_cores = <optimized out>
        pkg_data = <optimized out>
        core_data = <optimized out>
        cpu_data = <optimized out>
        buffer = "\000 61.8 ms", '\000' <repeats 103 times>, "\240\301I\000\000\000\000\000\307ӝ\367\377\177\000\000\377\377\377\377\377\377\377\377", '\000' <repeats 16 times>, "p˝\367\377\177\000\000\377\377\377\377ESSA\000\246v\315~\305r\357rtop.mo\000\000\246v\315~\305r\357\000\000\000\000\000\000\000\000\360\205"...
        buffer2 = "\000\071\070.3%\000\000\230ɉ\000\000\000\000\000\060\303\377\377\377\177\000\000\230ɉ\000\000\000\000\000\340\006\000\000\000\000\000\000\v\000\000\000\000\000\000\000\330Љ\000\000\000\000\000\336\071D\000\000\000\000\000\b\000\000\000\000\000\000\000\310\367\204\000\000\000\000\000\220ɉ\000\000\000\000\000\023\000\000\000\000\000\000\000\v", '\000' <repeats 15 times>, "p\301\377\377\001\000\000\000\300\367\204\000\000\000\000\000<\000\000\000\000\000\000\000\230ɉ\000\000\000\000\000\220\255O\000\000\000\000\000\023\000\000\000\000\000\000\000\367FF\000\000\000\000\000\a\206F\000\000\000\000\000\060\216F\000\000\000\000\000OGF\000\000\000\000\000>\216F\000\000\000\000\000OGF\000\000\000\000\000"...
        tmp_num = "5\000re 2\000\000\020*\205\000\000\000\000\000LC_MESSAGES/powe", '\000' <repeats 16 times>, "PA"
        package = <optimized out>
        core = <optimized out>
        cpu = <optimized out>
        line = <optimized out>
        cstates_num = <optimized out>
        title = <optimized out>
        core_num = <optimized out>
        _package = <optimized out>
        _core = <optimized out>
        _cpu = <optimized out>
        core_type = <optimized out>
        div_attr = {css_class = 0x4646f7 "clear_block", css_id = 0x4646ef "cpuidle"}
        std_table_css = {table_class = 0x7fffffffbcd0 "\340\274\377\377\377\177", td_class = 0x7ffff7fdc3b1 <_dl_lookup_symbol_x+289> "H\203\304\060\205\300t\267I\213M", tr_class = 0x7 <error: Cannot access memory at address 0x7>, th_class = 0x7ffff799d808 "\315\061@", pos_table_title = L, title_mod = 0, rows = 0, cols = 0}
        pkg_tbl_size = {rows = <optimized out>, cols = 2}
        core_tbl_size = {rows = <optimized out>, cols = 2}
        cpu_tbl_size = <optimized out>
        title_attr = {css_class = 0x468e30 "content_title", css_id = 0x46474f ""}
        idx1 = 8
        idx2 = <optimized out>
        idx3 = <optimized out>
        tmp_str = "  61.8 ms"
#5  0x0000000000412d6d in one_measurement (seconds=-4, seconds@entry=1, sample_interval=<optimized out>, sample_interval@entry=5, workload=workload@entry=0x0) at main.cpp:265
No locals.
#6  0x0000000000412e39 in make_report (time=20, workload=workload@entry=0x7fffffffd3e0 "", iterations=iterations@entry=1, sample_interval=sample_interval@entry=5, file=file@entry=0x7fffffffc3e0 "powertop.html") at main.cpp:300
No locals.
#7  0x00000000004097c6 in main (argc=2, argv=0x7fffffffe518) at main.cpp:521
        option_index = 5
        c = <optimized out>
        filename = "powertop.html", '\000' <repeats 2683 times>...
        workload = '\000' <repeats 4095 times>
        iterations = 1
        auto_tune = 0
        sample_interval = 5
```

jubalh, 2020-07-24:
Can you try whether the following patch makes a difference for you?

```
diff --git a/src/cpu/cpu.cpp b/src/cpu/cpu.cpp
index a92f111..dc383d9 100644
--- a/src/cpu/cpu.cpp
+++ b/src/cpu/cpu.cpp
@@ -561,14 +561,14 @@ void report_display_cpu_cstates(void)
 						core_type = _core->get_type();
 						if (core_type != NULL) {
 							if (strcmp(core_type, "Core") == 0 ) {
-								core_data[idx2]="";
+								core_data[idx2]=string("");
 								idx2+=1;
 								snprintf(tmp_num, sizeof(tmp_num), __("Core %d"), _core->get_number());
 								core_data[idx2]=string(tmp_num);
 								idx2+=1;
 								core_num+=1;
 							} else {
-								core_data[idx2]="";
+								core_data[idx2]=string("");
 								idx2+=1;
 								snprintf(tmp_num, sizeof(tmp_num), __("GPU %d"), _core->get_number());
 								core_data[idx2]=string(tmp_num);
@@ -770,7 +770,7 @@ void report_display_cpu_pstates(void)
 					buffer[0] = 0;
 					buffer2[0] = 0;
 					if (line == LEVEL_HEADER) {
-						core_data[idx2]="";
+						core_data[idx2]=string("");
 						idx2+=1;
 						snprintf(tmp_num, sizeof(tmp_num), __("Core %d"), _core->get_number());
 						core_data[idx2]=string(tmp_num);
```

You need to put this in a file `my.patch` and then do `patch -p1 < my.patch` in the main directory. Then rebuild and test.

Mershl, 2020-07-24:
The error changed. Interesting change:

        std_table_css = {table_class = 0xffffffff <error: Cannot access memory at address 0xffffffff>, td_class = 0x0, tr_class = 0x7ffff79af7c8 "\330S", th_class = 0xec0e3329 <error: Cannot access memory at address 0xec0e3329>, pos_table_title = 4160742568, title_mod = 32767, rows = 4205387, cols = 0}

Full stack trace attached: [2.txt](https://github.com/fenrus75/powertop/files/4971781/2.txt)

Mershl, 2020-07-26:
```
		core_tbl_size.rows=(cstates_num *_package->children.size())
				+ _package->children.size();
		core_tbl_size.rows+=_package->children.size();
```
The segmentation fault is fixed by adding an additional package to the core_tbl rows. I was able to reproduce this issue on a Ryzen 1600x and a Ryzen 3500U - potential misinterpretation for Ryzen CPUs?

jubalh, 2020-07-27:
When doing this  you still need my changes from above mentioned patch or it works without them?

Mershl, 2020-07-27:
It works without the string init patches. None the less the patch improves readability IMO.

The culprit are the parameters used for size in https://github.com/fenrus75/powertop/blob/master/src/cpu/cpu.cpp#L494 on a Ryzen CPU. The "hack" does not fix the underlying issue and will define the array (the rows of the table later) too big on previously working CPUs.

gkammela, 2020-07-27:
Thank you @Mershl and @jubalh for finding and root cause the issue. We will work on a fix. thanks!

AirDeeEx, 2020-07-30:
Hello, I have the same error.

PowerTOP version 2.14-pre
Kernel: 5.7.10-zen1-1-zen (Arch Linux)
Hardware: Asus VivoBook 15 X505ZA (AMD Ryzen 2500U)

Build parameters: CXXFLAGS='-g -Og' CFLAGS='-g -Og -pthread'
Run parameters: '-r --debug'

LLDB backtrace: [powertop_sigsegv_lldb.txt](https://github.com/fenrus75/powertop/files/5003821/powertop_sigsegv_lldb.txt)
GDB backtrace: [powertop_sigsegv_gdb.txt](https://github.com/fenrus75/powertop/files/5003820/powertop_sigsegv_gdb.txt)

evanstucker-hates-2fa, 2020-08-06:
I'm having the same issue.

powertop 2.13
System76 Serval WS with AMD Ryzen 5 3600 6-Core Processor

DawidLoubser, 2021-03-24:
Hi, I have the exact same issue on a Lenovo Thinkpad T495s (Ryzen 7 3700U)

imp1sh, 2021-08-14:
And another one on a Thinkpad T14 Ryzen 4650 Pro

xhoneybear, 2021-09-05:
another one there, HP Pavilion Aero on Ryzen 5 5600U with the same issue 🙋🏻‍♀️

MathewRomy, 2021-09-13:
I'm having the same issue on Acer Aspire Ryzen 5.

gvolt, 2021-09-14:
I discovered the same problem on a Gigabyte B550 AORUS ELITE V2 with a Ryzen 5 5600X.

Powertop: 2.14
OS: openSUSE Tumbleweed
Kernel: 5.14.x

With "Global C-state Control" disabled in the MB's firmware the segmentation fault does not occur (what was probably to be expected).

drishal, 2021-12-06:
> And another one on a Thinkpad T14 Ryzen 4750U Pro

same issue with thinkpad P14s ryzen 7 4750u

koriwi, 2022-01-21:
same issue with ideapad 5 15are05 ryzen 7 4800u

dirkmueller, 2022-04-14:
Any chance of a new release of powertop as the two commits have been merged into git `master` by now?

MatthewStadter, 2022-04-25:
Looking through the issues, I believe this issue is the same as https://github.com/fenrus75/powertop/issues/34.

I have this issue on a AWS EC2 m6a.xlarge (AMD).

DarwinSurvivor, 2022-07-25:
I think I'm having the same issue on my Ryzen 5 5600X desktop.

```
modprobe cpufreq_stats failedCannot load from file /var/cache/powertop/saved_results.powertop
Cannot load from file /var/cache/powertop/saved_parameters.powertop
File will be loaded after taking minimum number of measurement(s) with battery only 
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask 5
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask 5
Devfreq not enabled
glob returned GLOB_ABORTED
Cannot load from file /var/cache/powertop/saved_parameters.powertop
File will be loaded after taking minimum number of measurement(s) with battery only 
Preparing to take measurements
Segmentation fault
```

eckelj, 2022-08-29:
Hi, I hav the seg-fault iss on servers:

Packages:
	0: Intel Xeon Platinum 8259CL
Cores:
	0: 2 processors (0-1), Intel Sky Lake

dlime, 2022-11-28:
Same with a Thinkpad T14 (gen 3) AMD Ryzen 7 PRO 6850U

Happens with `--html` and `--csv`

demaniak, 2023-02-15:
Seems I hit the same problem.

`AMD Ryzen 7 5800H with Radeon Graphics`

the-mrd, 2023-02-23:
Hitting same thing on Ubuntu 22.04.2 LTS with AMD EPYC 7702. If needed, I can provide backtrace.

leon-thomm, 2024-01-12:
same on AMD Ryzen 7 7840U, running Kubuntu-22.04 on a Framework

oct1473254, 2024-04-26:
Had the same issue on POP_OS, lenovo thinkpad, T14s, powertop 2.14

Eric-Tyrrell22, 2025-05-04:
Upgrading powertop fixed report generation issues for me. Built v2.15 from source because the Jammy repo version (2.14.1) still had the problem.