fenrus75/powertop #65: Power usage estimate of 4.04 W of display backlight with turned of internal display

paulmenzel, 2020-07-28:
On a Dell Latitude E7250 with Debian Sid/unstable with Linux 5.7.6 and PowerTOP 2.11, connecting a monitor over HDMI and disabling the internal screen in GNOME 3.36.4, PowerTOP estimates four Watts of power usage for the display backlight with a utilization of zero percent.

      4.04 W      0,0%                      Device         Display backlight

Shouldn’t that be 0? I looked a little into the backlight code, but there have been no changes since a while.

    $ ls -l /sys/class/backlight/
    insgesamt 0
    lrwxrwxrwx 1 root root 0 28. Jul 13:43 intel_backlight -> ../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight

gkammela, 2020-09-24:
hi @paulmenzel, seems like a lot of issues with the power estimation feature. we are currently looking into this issue.

paulmenzel, 2024-02-21:
@gkammela, were you able to look into this issue.

Using Debian sid/unstable with *powertop* 2.15-3 on the different system Dell XPS 13 9360, the probem is still there with an external monitor connected over DisplayPort to a Dell DA300 USB-C adapter, and the internal display being turned off:

```
The battery reports a discharge rate of 17.9 W
The energy consumed was 154 J
The estimated remaining time is 0 hours, 40 minutes

Summary: 583.2 wakeups/second,  0.0 GPU ops/seconds, 0.0 VFS ops/sec and 34.7% CPU use

Power est.              Usage       Events/s    Category       Description
  1.38 W      0.0%                      Device         Display backlight
  813 mW     55.4%                      Device         USB device: USB 10/100/1000 LAN (Realtek)
  209 mW      8.9 pkts/s                Device         Network interface: enx18dbf22dccf3 (r8152)
```