fenrus75/powertop #69: The power usage estimation of a single item is higher than the total battery drain

ArthurBorsboom, 2020-09-23:
Is this a powertop, a driver (iwlwifi, battery, ... ) or another issue?
Where should I start diagnosing? :)

[image]

gkammela, 2020-09-24:
hi @ArthurBorsboom, this definitely seems like an issue with power estimation calculation. This feature was disabled for some time but we enabled it recently. We need to dig deep into the issue.

ArthurBorsboom, 2020-09-24:
Thanks for the clarification.

I am not sure if it is relevant, but I have noticed that the battery kernel module reports a warning on charger unplug about an inaccurate discharge rate.

```
battery: [Firmware Bug]: battery: (dis)charge rate invalid.
```

Please let me know if I need to provide information, such as the kernel log (dmesg), dmidecode, an ACPI dump, etc.