fenrus75/powertop #72: Cannot turn off "Bluetooth device" tuneable

WhyNotHugo, 2020-10-19:
I'm using powertop, and it's a bit aggressive suspending my Bluetooth, so my mouse is suspended if I stop moving it for a few seconds.

I'm trying to turn this off by running `powertop`, and then in the Tuneables tab I select `Bluetooth device interface status` and hit enter.

This seems to attempt to run `/usr/sbin/hciconfig hci0 up &> /dev/null &`. However, `hciconfig` has been deprecated upstream and no longer ships with bluez (it's been absent for about 3 years now).

Bluez seems to now include `btmgmt`, though I haven't been able to figure out how to turn off power saving with it.

Hikandy, 2021-09-20:
same error did you find anything yet?

seminarian, 2023-01-05:
FYI: My BT mouse was fixed after putting this line to bad: 

> >> Bad           Autosuspend for unknown USB device 3-10 (8087:0033)