fenrus75/powertop #74: gettext version on Ubuntu.

stolk, 2020-11-03:
I have an up to date Ubuntu LTS, and used the apt command from the documentation.

Why trying to build...

```
$ sh autogen.sh 
autoreconf: Entering directory `.'
autoreconf: running: autopoint
autopoint: *** The AM_GNU_GETTEXT_VERSION declaration in your configure.ac
               file requires the infrastructure from gettext-0.20.2 but this version
               is older. Please upgrade to gettext-0.20.2 or newer.
autopoint: *** Stop.
autoreconf: autopoint failed with exit status: 1
```

Is the version dictated by configure.ac unnecessary high, or does Ubuntu LTS just lag too much?

```
$ dpkg --list | grep libgettext
ii  libgettextpo-dev:amd64                     0.19.8.1-10build1                     amd64        process PO files - static libraries and headers
ii  libgettextpo0:amd64                        0.19.8.1-10build1                     amd64        process PO files - shared library
```

pannal, 2021-04-10:
Same issue here.

bartgrefte, 2021-05-05:
Same error in Debian 10.