fenrus75/powertop #76: running powertop drops all scaling_max_frequency to lowest on arm64 yoga c630 laptop

bm16ton, 2020-11-17:
Hello, On ubuntu 20.04 with powertop (from repo or newest git) when I run the program It sets my scaling_max_frequency to lowest possible frequency. My laptop has 4 slow cores at 1.7ghz and 4 fast cores at 2.9ghz. It seems mostly random on which cores it will make slow, usually (%65? of time) the slow cores get slowed, slightly less often the higher cores get slowed and sometimes both. I edited  src/cpu/abstract_cpu.cpp and commented out   lines 479 to 492 which are supposed to reapply correct max scaling freq after "wiggling" the cpu .Now the issue seems to have stopped. I can fix the cpu's by simply running;
echo 2956800 > /sys/devices/system/cpu/cpufreq/policy4/scaling_max_freq
echo 1766400 > /sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq

If this is caused by the HMP multicore setup or something else Im not sure. If a fix could be mainlined even if a command line or compile time option would be great. Thank you

56quarters, 2021-10-03:
I've seen similar behavior on a Lenovo Thinkpad X1 gen 9. Powertop sets the `scaling_max_frequency` to the lowest possible frequency on a single core.

It seems like this is possible if:
* You exit powertop at [just the right time](https://github.com/fenrus75/powertop/blob/2bcc17ace38d9bfea2e16f33192d786c4bf7db92/src/cpu/abstract_cpu.cpp#L483), before it resets the max frequency in `abstract_cpu::wiggle`
* You have two instances of powertop running at the same time, both trying to "wiggle" the same CPU core

I believe a partial fix here would be use a destructor of a some class to reset the max frequency when the `abstract_cpu::wiggle` method exits.

bm16ton, 2021-10-17:
I  totally forgot all about this, i apparently even posted my fork on my github
 https://github.com/bm16ton/yoga-c630-powertop 

But i have no idea what im.doing i just play around till it all seems to work. I just received a second yoga not long ago. Hopefully soon ill be able to look at this again and any changes made upstream since my last commit.

iskra-anl, 2023-08-30:
I'm seeing what appears to be the same problem on the Framework laptop with 13th gen Intel CPU (Raptor Lake, i5-1340p). Most frequently CPU 0 gets "stuck" at the lowest frequency and needs to be adjusted manually. I came to believe that it's powertop that causes it. I'm running version 2.15. This is a serious problem!

iskra-anl, 2023-08-30:
Also https://bugzilla.redhat.com/show_bug.cgi?id=1958449