fenrus75/powertop #81: Bogus (380 mW) value for power consumption

patrakov, 2021-03-14:
I have a HP EliteBook 735 G6 laptop. It is AMD-based. On it, powertop reports completely bogus values for power consumption and battery time remaining. Something like:

```
The battery reports a discharge rate of 368 mW
The energy consumed was 7.38 J
The estimated remaining time is 50 hours, 6 minutes
```

At the same time, Cinnamon battery applet correctly reports that the battery is good only for 2 hours and 20 minutes.

Presumably, this is because none of the methods that powertop can use for determining the battery discharge rate actually works:

```
20:06:58.602628 openat(AT_FDCWD, "/sys/class/power_supply/BAT0/present", O_RDONLY) = 123
20:06:58.602687 read(123, "1\n", 8191)  = 2
20:06:58.603210 openat(AT_FDCWD, "/sys/class/power_supply/BAT0/status", O_RDONLY) = 123
20:06:58.603286 read(123, "Discharging\n", 8191) = 12
20:06:58.604335 openat(AT_FDCWD, "/sys/class/power_supply/BAT0/power_now", O_RDONLY) = -1 ENOENT (No such file or directory)
20:06:58.604398 openat(AT_FDCWD, "/sys/class/power_supply/BAT0/energy_now", O_RDONLY) = -1 ENOENT (No such file or directory)
20:06:58.604449 openat(AT_FDCWD, "/sys/class/power_supply/BAT0/voltage_now", O_RDONLY) = 123
20:06:58.604500 read(123, "11140000\n", 8191) = 9
20:06:58.604571 openat(AT_FDCWD, "/sys/class/power_supply/BAT0/current_now", O_RDONLY) = 123
20:06:58.604620 read(123, 0x5630a5a7e730, 8191) = -1 ENODEV (No such device)
```

The only usable attribute is `charge_now`, but, alas, powertop does not know how to use it.

```
$ LANG=en_US.UTF-8 grep -rH .
uevent:POWER_SUPPLY_NAME=BAT0
uevent:POWER_SUPPLY_TYPE=Battery
uevent:POWER_SUPPLY_STATUS=Discharging
uevent:POWER_SUPPLY_PRESENT=1
uevent:POWER_SUPPLY_TECHNOLOGY=Li-ion
uevent:POWER_SUPPLY_CYCLE_COUNT=35
uevent:POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11550000
uevent:POWER_SUPPLY_VOLTAGE_NOW=11166000
uevent:POWER_SUPPLY_CHARGE_FULL_DESIGN=4330000
uevent:POWER_SUPPLY_CHARGE_FULL=3009000
uevent:POWER_SUPPLY_CHARGE_NOW=1459000
uevent:POWER_SUPPLY_CAPACITY=48
uevent:POWER_SUPPLY_CAPACITY_LEVEL=Normal
uevent:POWER_SUPPLY_MODEL_NAME=Primary
uevent:POWER_SUPPLY_MANUFACTURER=Hewlett-Packard
uevent:POWER_SUPPLY_SERIAL_NUMBER=13856 2020/04/22
charge_full_design:4330000
serial_number:13856 2020/04/22
technology:Li-ion
grep: current_now: No such device
charge_now:1459000
present:1
power/runtime_active_time:0
power/runtime_status:unsupported
grep: power/autosuspend_delay_ms: Input/output error
power/runtime_suspended_time:0
power/control:auto
manufacturer:Hewlett-Packard
hwmon2/in0_input:11166
hwmon2/power/runtime_active_time:0
hwmon2/power/runtime_status:unsupported
grep: hwmon2/power/autosuspend_delay_ms: Input/output error
hwmon2/power/runtime_suspended_time:0
hwmon2/power/control:auto
grep: hwmon2/curr1_input: No such device
hwmon2/name:BAT0
type:Battery
charge_full:3009000
capacity:48
cycle_count:35
voltage_now:11166000
status:Discharging
alarm:0
model_name:Primary
voltage_min_design:11550000
capacity_level:Normal
```

fenrus75, 2022-11-30:
> Same problem here with Fedora 37 and hp Elitebook 830 g5 , measuring around 360mW

this data comes from your bios ... not a lot powertop can do if the
bios lies to it

spbisc97, 2022-11-30:
It was strange for me because other utilities like the simple "power statistics" by gnome take the correct discharge rate.
Anyway thanks for your help and for the excellent utility!

paulmenzel, 2023-01-23:
Over what interface are these values read from the system firmware (“BIOS”)?