fenrus75/powertop #88: powertop aborted with *** stack smashing detected ***: terminated

DavidWHodgins, 2021-07-08:
Running PowerTOP version 2.13 on Mageia 8 x86_64. If a user is running the NetBeans IDE and root runs powertop, then
powertop aborts with stack smashing detected. Usually the initial display is ok, and then it aborts on the first refresh, but
sometimes it aborts on the first display.
[powertop-backtrace.txt](https://github.com/fenrus75/powertop/files/6780675/powertop-backtrace.txt)

DavidWHodgins, 2021-07-08:
Just fyi, if I start powertop first and after a couple of refreshes start the NetBeans IDE, powertop aborts on the next refresh.

DavidWHodgins, 2021-07-08:
Mageia bug report is at https://bugs.mageia.org/show_bug.cgi?id=29198

MichaelDietzel, 2022-04-21:
I have the same issue under kubuntu 21.10 running powertop 2.11 but without running java. I'll try to to some tests if I can find any program triggering the crash.

jlaunonen, 2023-01-25:
I believe this might be fixed by commit 33a23e7517c1b2e4e6729feb9bf90fd937d797ce that is present in tag v2.15
A way to test the buffer overflow is to add `-fsanitize=address -ggdb` compiler flags, like

```bash
export CFLAGS="-fsanitize=address -ggdb"
export CXXFLAGS="${CFLAGS}"
./configure --prefix="$PWD/_out"
```

For the test, a command with a long cmdline (over 4096 bytes) is needed to be present in process table.
The custom powertop executable should then be run like

```bash
"$PWD/_out/sbin/powertop" 2>asan-log || reset
```

which should crash quite instantly, producing an asan-log file with stacktrace that is pointing to cmdline_to_string.
At beforementioned commit or at tag v2.15 the crash doesn't happen any more.

To create a controlled long cmdline, following might work, but I currently cannot test it as I have other processes that already have a long cmdline.

   bash -c 'echo "I am PID $$, hit ctrl-c to stop"; read </dev/zero >/dev/null #'$(seq -s " " -w 1000)

(That should cause one CPU core to be hogged for a busy work while all being in the same process.)

LvargaDS, 2023-12-15:
PowerTOP 2.14, 22.04.3 LTS (Jammy Jellyfish), same problem.

I have tried to install PowerTOP (to use version which should have problem fixed) from this repository. It is failing on `./configure` command with last line: `configure: error: libtracefs is required but was not found`, despite that `apt install libtracefs-dev` finished successfully.