fenrus75/powertop #101: Keyboard and mouse latency after running `powertop --auto-tune`

JaapJoris, 2022-06-14:
I am experiencing extreme keyboard and mouse latency after running `powertop --auto-tune`. Both input devices go to sleep after about 3 seconds, and take a few seconds to wake up again, resulting in missed keypresses and an unresponsive interface.

### Steps to reproduce

    powertop --auto-tune

### Expected result

The keyboard and mouse work normally

### Actual result

After ±3 seconds of inactivity the keyboard and mouse stop working. When you press a key, it isn't registered. After the initial keypres, this situation lasts for a few seconds until the keyboard responds normally again.

### Hardware

Logitech K400 Plus keyboard with integrated trackpad, which works correctly before running `powertop --auto-tune`. Here is the output of `xinput list`:

```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech K400 Plus                      	id=9	[slave  pointer  (2)]
⎜   ↳ Synaptics TM2668-001                    	id=11	[slave  pointer  (2)]
⎜   ↳ TPPS/2 IBM TrackPoint                   	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=10	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=13	[slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                  	id=14	[slave  keyboard (3)]
    ↳ Logitech K400 Plus                      	id=15	[slave  keyboard (3)]

```

DDvO, 2023-12-25:
> Noticed the same issue with Logitech, but i thought it was normal.

I experience the same issue, using powertop 2.14 on Ubuntu 22.04.
This cannot be normal - it makes the machine pretty unusable.
Has this been tackled/fixed meanwhile?

DDvO, 2023-12-26:
It turns out that this bug is more than 8 years old :-(
https://askubuntu.com/questions/678779/powertop-auto-tune-without-messing-with-usb-and-touchpad/1026527

That page contains several useful workarounds. I chose this one:
```
HIDDEVICES=$(ls /sys/bus/usb/drivers/usbhid | grep -oE '^[0-9]+-[0-9\.]+' | sort -u)
for i in $HIDDEVICES; do
  echo -n "Enabling " | cat - /sys/bus/usb/devices/$i/product
  echo 'on' > /sys/bus/usb/devices/$i/power/control
done
```
which disables power management for all Human Interface Device USB drivers.

Pointedstick, 2025-08-24:
Slightly a duplicate of https://github.com/fenrus75/powertop/issues/38.