fenrus75/powertop #105: How to read powertop output? It says my device usage is 100%

yurii-anikin, 2022-08-04:
Hello. First of all I am sorry for silly name of the issue.

My issue is that powertop shows that my built-in fingerprint reader usage is 100% and that is scary

[image]
Does it mean that my built-in fingerprint reader (USB device: usb-device-06cb-00fc) takes ... 100% of power? How to read that? I am sorry I am little bit stressed cause my thinkpad is very hot during just regular browser surfing....

another screenshot from powertop:
[image]

fenrus75, 2022-08-04:
it means that this device does not participate in any power saving and also
(as a USB device that is active) will keep your CPU from using its lowest
power modes

nikitosiusis, 2022-08-09:
And what can cause this behaviour?

nikitosiusis, 2022-08-09:
Yes, I understand this, but where should I look into, to debug which one is doing this?:)

st4ycool, 2022-08-10:
@fenrus75 thanks for answers. I googled more and found thinkpad users who are disabling fingerprint reader in bios, as well as other stuff like lan port, wan port(i dont have then physically, but they were in bios).
So, i guess its solver.

Very big thanks to powertop, without it I could not identify the problem!

@nikitosiusis into it's id?

yurii-anikin, 2022-08-13:
@fenrus75 thanks for answers. I googled more and found advices from thinkpad users who are disabling fingerprint reader in bios, as well as other stuff like LAN port, WAN port(i dont have then physically, but they were in bios). So did I.

But now I have other problem: 
Does it really means that my integrated webcamera dosn't participate in any power saving protocols? How can it be (

fenrus75, 2022-10-11:
if you have the "lsof" tool, it often is useful in finding which things
keep what files open, you'll be looking for something keep usb devices or
input devices open