fenrus75/powertop #107: Way off power estimation Intel AX201 Bluetooth, with HID devices active

shoffmeister, 2022-09-03:
Running `powertop` 2.14 on Fedora 36 (kernel 5.19.4-200) while on AC, and with two Bluetooth devices connected, I get
```
Power est.              Usage       Events/s    Category       Description
  49.3 W     94.7%                      Device         USB device: usb-device-8087-0026
  5.79 W    100.0%                      Device         Radio device: btusb
```
This data does not make any sense.

The USB device is
```
Bus 003 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth
```
which is a "Intel® Integrated Connectivity (CNVi)" piece, with what I would call "cost optimizations" in place to utilize main CPU capacity for handling Bluetooth and WiFi.

Connected are two Bluetooth input devices
* Logitech MX Keys Mini
* Logitech MX Master 3S

It would seem as if any activity on the Bluetooth devices - i.e. typing on the keyboard, or wiggling the mouse - causes the power estimate to skyrocket.

System is a Dell Inspiron 7610 (with Nvidia RTX 3060).

gkammela, 2022-09-29:
hi @shoffmeister the power estimate feature has been disabled because of the known issues.

paulmenzel, 2023-01-23:
Interesting. Can you please reference the commit disabling the feature?