fenrus75/powertop #109: Freeze when running powertop or powertop --calibrate

aacebedo, 2022-10-16:
Hi

I updated my kernel to try the incoming 6.1-rc1 (https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=1501278bb7ba0728b869d3399ea94b67853256a2).
when I start `powertop`, my laptop freezes (I do not see any feedback hinting to a kernel panic or whatsoever).
Running `powertop --debug` does not trigger the freeze.

What's the best way to debug this issue ?

aacebedo, 2022-10-17:
I retried on official 6.1-rc1 and saw the same behaviour

aacebedo, 2022-10-18:
What's your machine ? 
Mine is a ROG Zephyrus G14 GA402.
There are some modifications on the 6.1-rc1 kernel for this specific model (I was looking for those TBH) so it would be helpful to know if it's coming from the machine; from the kernel or from powertop.

aacebedo, 2022-10-18:
Note: I also tried with the official 6.1-rc1 tag and I got the same behaviour

rottenpants466, 2022-10-18:
My machine is a MSI Summit E16 Flip(Model: A12UCT), Alder Lake cpu. I'm running kubuntu 22.10 with the kernel updated to 6.1 rc.

Installs fine but when i try to run the powertop command the laptop just freezes

aacebedo, 2022-10-18:
exacte

> My machine is a MSI Summit E16 Flip(Model: A12UCT), Alder Lake cpu. I'm running kubuntu 22.10 with the kernel updated to 6.1 rc.
> 
> Installs fine but when i try to run the powertop command the laptop just freezes
> 
> Edit: powertop --debug command runs fine

Exact same behavior on NixOS with a custom modification to use the 6.1-rc1 (not integrated in repos yet)

`--debug` also works.

B83C, 2022-10-22:
Same issue here on T14s Gen 3 AMD, Ryzen 6850u, was running fine a few kernel tags before. The same happens when I run radeontop, no idea why...

and also, --debug works just fine as well

thor2002ro, 2022-10-25:
false alarm rc2 works fine

aacebedo, 2022-10-25:
Tried it yesterday and it still crashes.

thor2002ro, 2022-10-25:
works here... i built my 6.1-rc2 from git://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git master branch 
it usually has a little extra patches than the rc..... check it out

aacebedo, 2022-10-25:
I used the tag. Can you give me the commit you used on master ?

aacebedo, 2022-10-25:
tried https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=1a2dcbdde82e3a5f1db9b2f4c48aa1aeba534fb2 still crashing.

thor2002ro, 2022-10-25:
i built it with gcc and freezing is back.... lol so its something compiler related....

knurd, 2022-10-25:
Anyone tried to bisect this and report this upstream (e.g. to the Linux developers). I'm doing this now, but I wonder if maybe someone else did that already.

diegomontoya, 2022-10-25:
Fedora 38 rawhide with 6.1.0-0.rc1.20221019gitaae703b02f92.17.fc38 on amd zen3 6850h HP laptop has exact same issue. Instant lockup with blank screen when manually calling powertop.

aacebedo, 2022-10-25:
> Anyone tried to bisect this and report this upstream (e.g. to the Linux developers). I'm doing this now, but I wonder if maybe someone else did that already.

I did not. 
I started to see it 1 or 2  commits before rc1. It was here for sure at rc1 tag.

knurd, 2022-10-25:
# first bad commit: [838d9bb62d132ec3baf1b5aba2e95ef9a7a9a3cd] perf: Use sample_flags for raw_data

Reverting that commit in master fixes the issue for me. Applying https://lore.kernel.org/lkml/20221020070657.21571-2-atrajeev@linux.vnet.ibm.com/ does not.

knurd, 2022-10-25:
This fixed it for me:
https://lore.kernel.org/all/20221012143857.48198-1-james.clark@arm.com/

aacebedo, 2022-10-26:
Do you know if the commit you mentioned is going to be integrated in the master? It says not reachable from any branch for the moment.

knurd, 2022-10-26:
> Do you know if the commit you mentioned is going to be integrated in the master?

You mean this fix? https://lore.kernel.org/all/20221012143857.48198-1-james.clark@arm.com/ Yes, it'll likely be merged this week.

aacebedo, 2022-10-26:
great !
It strongly impact the use of my machine as I see big battery life differences when I am not running the auto-tune feature of powertop. I tried power-profile-daemon and tlp but it is not as efficient.

thor2002ro, 2022-10-26:
the interesting part is why does it work when compiled with clang..... :))))

aacebedo, 2022-10-29:
Tried HEAD of next (https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/snapshot/linux-next-fd8dab197cca2746e1fcd399a218eec5164726d4.tar.gz) and it solves the problem.
I checked that the lines modified by the patch mentioned by @knurd are in this branch. 
Now just have to wait for this to be upstreamed.

I guess I shall close this issue as it is not related to powertop.

knurd, 2022-10-29:
> Now just have to wait for this to be upstreamed.

I expect that to happen tomorrow, e.g. before 6.1-rc3 is released

aacebedo, 2022-10-29:
Cool. I'll wait for the tag then and update the related files in nixpkgs to point on it.

paulmenzel, 2023-01-23:
@aacebedo, any updates? Is the issue fixed, so it can be closed?

aacebedo, 2023-01-23:
Yes it is fixed

closed 2023-01-23 by aacebedo