fenrus75/powertop #111: calibration: modprobe cpufreq_stats failed

t-nil, 2022-10-17:
First line of calibration output is `modprobe cpufreq_stats failed`. Neither my arch nor my fedora installation seem to have that kernel module (`/lib/modules/…`). Is it still sensible to rely on that module?

LinuxOnTheDesktop, 2023-08-07:
I see the same message - on Mint (and plus - cf. #128 - I see `Devfreq not enabled`).

A post on [this webpage](https://askubuntu.com/questions/1131279/powertop-cpufreq-stats-failed-and-devfreq-not-enabled-running-auto-tone) says that (repeated) calibration suppreses the modprobe warning (and the other warning), but it does not do so for me - presumably (in the case of the modprobe warning) because, like t-nil, my system lacks the `cpufreq_Stats` module. Certainly I cannot find on my system a file of that name.

NXTler, 2023-10-25:
I have the same issue, has there been any solution to this?