fenrus75/powertop #114: missing power estimation ?

pc00per, 2022-10-24:
I've just installed it on arch & the overview doesn't seem to show power est. beside usage ? Documentation & manpages doesn't seem to explain about this too.

ansoniikun, 2024-02-05:
did you ever find a solution to this?

fenrus75, 2024-02-05:
to get power estimate one first needs quite a bit of time with powertop
running while on battery (or some other setup where it can read actual
power) to train the model

without  that it won't know what to estimate

rajeev1976, 2024-08-19:
> to get power estimate one first needs quite a bit of time with powertop running while on battery (or some other setup where it can read actual power) to train the model without that it won't know what to estimate
> […](#)

Hi,
     What if i have to run powertop with system without battery(like desktop or servers), how do we get the power data on these systems ?

thanks
Rajeev