fenrus75/powertop #123: inaccurate Pkg(HW) percentages on Rocket Lake v2.15

TheLinuxGuy, 2023-04-16:
Hello, I think I may have found a bug in the latest release, the percentages are off and do not total 100% when aggregated together.

Example from the screenshot "C2 (pc2) 23.8%" all other states are 0.0% and so there seems to be missing or miscalculated 76.2% that is missing from stats view.
[image]

powertop and sysinfo
`# lshw -C CPU
  *-cpu
       description: CPU
       product: 13th Gen Intel(R) Core(TM) i5-13600K
       vendor: Intel Corp.
       physical id: 25
       bus info: cpu@0
       version: 13th Gen Intel(R) Core(TM) i5-13600K
       serial: To Be Filled By O.E.M.
       slot: CPUSocket
       size: 801MHz
       capacity: 5100MHz
       width: 64 bits
       clock: 100MHz
       capabilities: lm fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp x86-64 constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req hfi umip pku ospke waitpkg gfni vaes vpclmulqdq tme rdpid movdiri movdir64b fsrm md_clear serialize pconfig arch_lbr ibt flush_l1d arch_capabilities cpufreq
       configuration: cores=14 enabledcores=14 threads=20
root@centrix:~# powertop -V
PowerTOP version 2.15
`

fenrus75, 2023-04-16:
> enabledcores=14 threads=20 ***@***.***:~# powertop -V PowerTOP version

there is no way to measure PC0/PC1... so while this could be a bug .. it's
more likely that something is keeping your SOC awake a lot of the time....

TheLinuxGuy, 2023-04-18:
> there is no way to measure PC0/PC1... so while this could be a bug .. it's more likely that something is keeping your SOC awake a lot of the time....
> […](#)

I see, thanks for clarifying. It's highly likely related to PCIE 3.0 X1 SATA controller ASM1064 on my system.