fenrus75/powertop #124: Build error: configure: error: libtracefs is required but was not found

vrossum, 2023-04-24:
I get a build error on Ubuntu22.04
configure: error: libtracefs is required but was not found

> apt install -y libtracefs1
returns already installed

fenrus75, 2023-04-24:
you're going to need to install the -dev version of that package, not just
the library

vrossum, 2023-04-24:
You are right. But it is already installed as well.

apt install -y libtracefs-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libtracefs-dev is already the newest version (1.2.5-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

yeongjoshua, 2023-05-02:
This fix for me

```
git clone https://git.kernel.org/pub/scm/libs/libtrace/libtraceevent.git/
cd libtraceevent; make; sudo make install; cd ..;

git clone https://git.kernel.org/pub/scm/libs/libtrace/libtracefs.git/
cd libtracefs; make; sudo make install; cd ..;
```

ivan-koryshkin, 2023-05-04:
Same problem

`configure: error: libtracefs is required but was not found`

I have installed `libtracefs-dev` and `libtracefs1` in my system

stephematician, 2023-07-28:
I had to install `libtracefs1`, `libtracefs-dev`, `libtraceevent1` and `libtraceevent-dev`. The former two were insufficient on their own.

xmixahlx, 2024-07-31:
FYI resolved with installing libtracefs-dev and libtraceevent-dev on Debian Unstable.

Hoernchen, 2024-12-11:
Same issue on ubuntu 24.04, fixed by installing libtraceevent-dev.

ValeZAA, 2025-03-29:
Yep, Debian testing, no issue with tracefs_event_file_exists, just install libtraceevent-dev

This is not mentioned in Readme though under

Example packages to install in Ubuntu*: