fenrus75/powertop #139: https://www.01.org/powertop doesn't go anywhere useful any more

Daniel15, 2023-10-30:
https://www.01.org/powertop just goes to a landing page about Intel open-source. Can this be modified to go to something Powertop-specific? I was looking for more detailed documentation and was hoping the URL would go to it.

bogardon, 2024-01-11:
even searching "powertop documentation" doesn't yield any results that look official. Are there official docs? Even `man powertop`  has this line at the bottom:

> The program is more fully described at ⟨https://01.org/powertop⟩

danielhoherd, 2024-02-05:
IMHO this is the compelling reason why documentation belongs in the git repo along with its build and deploy scripts.

jubalh, 2024-02-05:
It is ;)
https://github.com/fenrus75/powertop/blob/master/doc/powertop.8

danielhoherd, 2024-02-05:
@jubalh https://github.com/fenrus75/powertop/blob/master/doc/powertop.8#L92-L93 disagrees with you.

jubalh, 2024-02-05:
If I remember correctly there was never any useful information at that adress.

danielhoherd, 2024-02-05:
That site used to link to this long PDF which does not appear in any GH searches, though I suppose it could be in a non-default branch like gh-pages or something. https://web.archive.org/web/20191019015936/https://01.org/sites/default/files/page/powertop_users_guide_201412.pdf

laama1, 2024-10-12:
ddg search still shows 01.org and wikipedia in the fisrt info panel / search hit. dangerously misleading