fenrus75/powertop #140: What does _ACPI suffix mean?

Daniel15, 2023-10-30:
One system gives me output like this:
```

           Pkg(HW)  |            Core(HW) |            CPU(OS) 0
                    |                     | C0 active   2.8%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          1.1%    0.4 ms
C2 (pc2)    7.2%    |                     |
C3 (pc3)    5.5%    | C3 (cc3)    0.0%    | C3          0.1%    0.1 ms
C6 (pc6)    1.5%    | C6 (cc6)    1.9%    | C6          2.2%    0.6 ms
C7 (pc7)   75.2%    | C7 (cc7)   92.8%    | C7s         0.0%    0.0 ms
C8 (pc8)    0.0%    |                     | C8         21.5%    2.5 ms
C9 (pc9)    0.0%    |                     | C9          0.0%    0.0 ms
C10 (pc10)  0.0%    |                     |
                    |                     | C10        72.8%   12.5 ms
                    |                     | C1E         0.4%    0.2 ms

                    |            Core(HW) |            CPU(OS) 1
```

while another gives output like this:
```
           Pkg(HW)  |            Core(HW) |            CPU(OS) 0   CPU(OS) 1
                    |                     | C0 active   5.9%        0.9%
                    |                     | POLL        0.1%    0.0 ms  0.0%    0.0 ms
                    |                     | C1_ACPI    14.2%    0.2 ms  1.0%    0.1 ms
C2 (pc2)    0.0%    |                     | C2_ACPI    39.2%    0.8 ms 27.0%    0.9 ms
C3 (pc3)    0.0%    | C3 (cc3)    0.0%    | C3_ACPI    33.6%    1.2 ms 69.7%    3.0 ms
C6 (pc6)    0.0%    | C6 (cc6)    1.1%    |
C7 (pc7)    0.0%    | C7 (cc7)    0.0%    |
C8 (pc8)    0.0%    |                     |
C9 (pc9)    0.0%    |                     |
C10 (pc10)  0.0%    |                     |

                    |            Core(HW) |            CPU(OS) 2   CPU(OS) 3
```

What's the difference between "C3" and "C3_ACPI", and why would the second system have fewer states? (first one is an i5-9500 while the second one is an i5-13500, both with C-states enabled in the BIOS)

fenrus75, 2023-10-30:
it means your kernel does not have intel_idle support for your CPU ...
which is unusually unless you're running an older kernel on newer hardware

Daniel15, 2023-10-30:
Hmm, that's interesting, because it does seem like `intel_idle` is active?

```
root@DanHome:~# cat /sys/devices/system/cpu/cpuidle/current_driver
intel_idle
```

It's an i5-13500 running on Linux 6.1.38. I'll have to figure out how to investigate this more. Can any of the `cpufreq` settings affect this?

fenrus75, 2023-10-30:
intel_idle can fall back to ACPI -- but that does not mean it knows your
hardware.
Is this by chance new hardware ? (newer than the 6.1 kernel which is
basically a year old by now)

Daniel15, 2024-01-14:
Hey @fenrus75, sorry for the delay in replying. This is somewhat recent hardware - it's a W680 motherboard with a Raptor Lake CPU (i5-13500).

I did notice that `intel_idle.c` seems to have cases for `ALDERLAKE` and `ALDERLAKE_L` but not `RAPTORLAKE`: https://github.com/torvalds/linux/blob/052d534373b7ed33712a63d5e17b2b6cdbce84fd/drivers/idle/intel_idle.c#L1530

Daniel15, 2024-01-23:
Confirmed that intel_idle doesn't currently support Raptor Lake CPUs, as per this mailing list post from August 2023: https://lore.kernel.org/linux-pm/7e2c1da24b48217045e8ad95b739ec96cdce5931.camel@intel.com/