fenrus75/powertop #143: powertop issues with i226

decommentator, 2023-12-11:
when having multiple i226 integrated in one box I see the following strange behaviour

clean install with proxmox
PowerTOP version 2.14
running powertop no problem everything works as should be

running powertop --auto-tune
afterwards:
not possible to run powertop anymore 
not possible to reboot

tomillr, 2024-05-19:
I can confirm this issue when running "powertop --auto-tune" on an Odroid H4+ board which use two Intel I226-V network ports.

[Odroid H4 hardware specification](https://wiki.odroid.com/odroid-h4/hardware)

fenrus75, 2024-05-19:
this is a device driver bug... it fails when turning on power management

tomillr, 2024-05-19:
Thanks for your assessment. 
I'm currently running Ubuntu Server 24.04. with just the basic kernel drivers. Is there maybe a proprietary Intel driver that properly supports power management?

fenrus75, 2024-05-19:
Intel doesn't do proprietary drivers... but you can file a bug against the
driver at bugzilla.kernel.org and it'll get to the maintainers

tomillr, 2024-05-19:
How about this driver for the I226-V from the official Intel website?

https://www.intel.com/content/www/us/en/download/15084/intel-ethernet-adapter-complete-driver-pack.html

tomillr, 2024-09-01:
@fenrus75 
Could you maybe chime in over at Bugzilla? You're input surely carries more weight and might help to get this bug fixed.

https://bugzilla.kernel.org/show_bug.cgi?id=218499

popy2k14, 2025-03-21:
Kernel 6.12 (new LTS) dropped a few months back.
I read somewhere that the issue was solved around ~6.11 kernel.

Can anyone of you please confirm the issue is solved in 6.12?

tomillr, 2025-03-23:
> Can anyone of you please confirm the issue is solved in 6.12?

I can confirm that the problem seems to have been fixed with Kernel 6.11 (tested with Ubuntu Server 24.10).

popy2k14, 2025-03-23:
thx for confirming!

Will get an I226-V now for my Unraid server and hopes it runs stable with ASPM (power savings turned on) and i can reach higher c-states without the need to disable ASPM.

Will report back here after i have tested it.

popy2k14, 2025-04-05:
I have now tested the I226-V in an Gigabyte H610i Mainboard with the latest F33 BIOS.
Had to disable "Plattform Power Management -> PEG ASPM" (giving up all cpu pkg c states) to disable ASPM fo the card to get it stable.

With "PEG ASPM" enabled (ASPM enabled for the NIC card) it was functioning and not freeze the system (so above issue seems to be solved), but the speed was dropping under 100 Mbits/s often. Also SMB file transfer to unraid was very unstable!

Maybe ill try an I226-T1 again.

Herve-M, 2025-07-13:
> > Can anyone of you please confirm the issue is solved in 6.12?
> 
> I can confirm that the problem seems to have been fixed with Kernel 6.11 (tested with Ubuntu Server 24.10 running on a Odroid H4+).

Which version of kernel are you using? My Ubuntu 24.04  using `6.11.0-29-generic` still have the problem.

popy2k14, 2025-09-29:
My workaround hasn't changed, i disabled ASPM for the card to keep bandwidth up.

popy2k14, 2025-09-29:
Sadly i am not aware of one.
I wand also enable "PEG ASPM"...
It seems an issue with the card/firmware itself and not linux related, see here some windows users on an NUC with the same NIC: https://www.reddit.com/r/intelnuc/comments/1951rgk/very_slow_speeds_with_i226v_on_intel_nuc13anbi5/

Dont know (or maybe) there is an updateable firmware on the card?