fenrus75/powertop #154: Allow running as non-root

EfogDev, 2024-08-07:
Hey, so it actually works (partially) just with `fakeroot`:
[image]

I believe it makes sense to allow running it as non-root? 
You don't get everything, but power consumption is there (the thing I personally need most of the time).

fenrus75, 2024-08-07:
hmmmm there really isn't a lot to report since almost all information is
privileged
(so some of what it reports is faked I suspect)

EfogDev, 2024-08-08:
Well my use case is I'm making a small GNOME extension to monitor momentary power consumption (running with `-t 1`). The readings look pretty accurate and consistent, except that (I think) it doesn't account for the screen in calculations. I'm pretty happy with the result, so it is better to get some data than to not get any at all, isnt't it?

fenrus75, 2024-08-08:
I'll ponder it some more... can you say exactly what data you want to use?