fenrus75/powertop #161: Idle stats

Solverz-0, 2024-10-30:
In the "Idle stats" tab, I see pkg(OS) instead of pkg(HW).
After some searching around the Web, I cannot find anyone else who sees OS instead of HW, do you know why I see this or how to find out why?

fenrus75, 2024-10-30:
well it means that powertop is using an OS/calculated value for this,
rather than a direct hardware reported value.
that by itself is not an issue, but it's slightly less accurate

what kind of cpu do you have? (not all cpus support reporting this directly)

Solverz-0, 2024-10-30:
Ah I see. I have an Intel N100.

The reason I ask is because I only see:
C2, C9 and C10 states (from what I remember, will double check soon).

So I thought something odd is going on, and my system seems to be in C0 or C10, nothing in between.

fenrus75, 2024-10-30:
it's quite common to see almost all time to be in the lowest C state (eg
C10) -- the OS often has very little reason to pick stuff inbetween.
(unless you're doing things like media playback or other very periodic low
latency things -- that's where the middle ones shine)

Solverz-0, 2024-10-30:
Okay that makes sense.

But I should see the middle c states right? They are simply not visible in powertop for me. Only the mentions ones are.

fenrus75, 2024-10-30:
not all numbers tend to be there -- usually there's about 3 total.

Solverz-0, 2024-10-30:
Hmm, I see. That lines up with what I saw. (again will double check later) 

But others with the same motherbaord and CPU and BIOs settings seem to see C2 -> C10 and Pkg(HW).
This is why I am confused.

fenrus75, 2024-10-30:
same OS kernel as well? but yeah that is a bit weirder

Solverz-0, 2024-10-30:
Yes, 6.1.

Looks like the powertop version in main for debian 12.7 is 2.14, not 2.15.
Maybe this is the difference? Can test.

z411, 2025-05-31:
Hi @Solverz-0 just wanted to say that the version is the reason. I compiled powertop 2.15 on Debian and it shows the HW reportes C-states now. You should do it because OS reported C-states on this CPU aren't accurate at all (in my case it showed C10 OS but the HW wasn't hitting below C3 at all).