fenrus75/powertop #165: Powertop autotune messes up the internal keyboard which is connected by USB

iopq, 2024-11-10:
https://community.frame.work/t/anyone-else-have-the-first-keyboard-input-after-the-touchpad-not-register/59532/

In tunables, powertop would tune the entry `Autosuspend for USB device Laptop 16 Keyboard Module - ANSI [Framework]` which would make it go to sleep after I touch the touchpad a little which means I had to press it to wake it back up, making the first press of the keyboard not register

christianhorn, 2025-01-07:
I guess this could be solved in 2 ways for you,

- having an "exclude list" for powertop autotune
- or modifying the keyboard driver to detect it's set to autosuspend and deal with it better.

Following up on the second path sounds more promising, might also lead to more people using autosuspend for the usb connected keyboards, so more power savings. You might want to bring this to the attention of the maintainers of the keyboard driver.

Pointedstick, 2025-08-24:
Duplicate of https://github.com/fenrus75/powertop/issues/101.