fenrus75/powertop #166: Ubuntu Server with APC UPS: always 0 prior measurements

moutasem1989, 2024-11-25:
In my setup I am running a headless Ubuntu server 24.04 LTS connected to an APC UPS. Trying to run Auto Tune returns:
```
$ sudo powertop --auto-tune
modprobe cpufreq_stats failed
Loaded 0 prior measurements
Cannot load from file /var/cache/powertop/saved_parameters.powertop
File will be loaded after taking minimum number of measurement(s) with battery only 
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask d
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask d
Devfreq not enabled
glob returned GLOB_ABORTED
Cannot load from file /var/cache/powertop/saved_parameters.powertop
File will be loaded after taking minimum number of measurement(s) with battery only 
Leaving PowerTOP
```
I tried calibrating and rebooting multiple times but `Loaded 0 prior measurements` remains.
It also promotes for `taking minimum number of measurement(s) with battery only` but it is unclear on how to do so.

any help is greatly appreciated!

jpa77, 2025-07-18:
Same problem here on [Proxmox VE ](proxmox-ve: 8.4.0 (running kernel: 6.8.12-11-pve))

meminens, 2025-09-21:
Any update on this?