fenrus75/powertop #176: Error during autogen.sh

Leitner1, 2025-03-25:
Hello,

I get an error when I try to run ./autogen.sh.

`autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: running: autopoint
autopoint: File po/Makefile.in.in has been locally modified.
autopoint: *** Some files have been locally modified. Not overwriting them because --force has not been specified. For your convenience, you find the local modifications in the file '/tmp/arXyOQCd/gthdLRIz/autopoint.diff'.
autopoint: *** Stop.
autoreconf: error: autopoint failed with exit status: 1
`

My system is:
Proxmox VE 8.3.5
CPU: 4 x Intel(R) Core(TM) i3-8100 CPU @ 3.60GHz (1 Socket)
Kernel: Linux 6.8.12-8-pve (2025-01-24T12:32Z)

Any idea?

r0bb10, 2025-03-25:
add --force to autogen.sh and it will go.

r0bb10, 2025-03-25:
> Oh sorry, I forgot to mention this, I already tried `./autogen.sh --force`, unfortunately it doesn't work, same error message. And `/tmp` folder is empty...

edit the sh autogen script and add --force after --verbose, save exit and relaunch

closed 2025-03-25 by Leitner1