fenrus75/powertop #177: Wrong CPU detected since debian update

Leitner1, 2025-04-08:
Hello everyone,

I set up my system about half a year ago with Proxmox VE 8.2.2 (Debian Linux 6.8.4-2-pve) and Powertop 2.14.
I got to package state C10 with -autotune, and everything was fine.

Since an `apt get update && apt get upgrade `a few days ago, I can only get to package state C8.

I noticed that Powertop now detects a different CPU under 'Device State'.
I have an Intel Core i3-8100 Coffee Lake CPU. Powertop, however, now shows "Cannon Lake"
Before that, "Coffee Lake" was actually shown, I'm sure of that.

Under Tunables everything  is set to Good.

Is there any way to assign the CPU manually?

I've already tried downgrading the kernel to 6.8.4, but unfortunately that didn't work.

Anyone have any ideas?

inxi -b
```
System:
  Host: pve Kernel: 6.8.12-9-pve arch: x86_64 bits: 64 Console: pty pts/0 Distro: Debian
    GNU/Linux 12 (bookworm)
Machine:
  Type: Desktop System: FUJITSU product: ESPRIMO D738 v: N/A serial: YMLJ008430
  Mobo: FUJITSU model: D3601-A1 v: S26361-D3601-A1 serial: xxxxxxxx UEFI: FUJITSU // American
    Megatrends v: 5.0.0.13 R1.32.0 for D3601-A1x date: 08/25/2023
CPU:
  Info: quad core Intel Core i3-8100 [MCP] speed (MHz): avg: 1500 min/max: 800/3600
Graphics:
  Device-1: Intel CoffeeLake-S GT2 [UHD Graphics 630] driver: i915 v: kernel
  Display: server: No display server data found. Headless machine? tty: 166x39
  API: OpenGL Message: GL data unavailable in console for root.
Network:
  Device-1: Intel Ethernet I219-V driver: e1000e
Drives:
  Local Storage: total: 2.77 TiB used: 12.36 GiB (0.4%)
Info:
  Processes: 322 Uptime: 3d 22h 17m Memory: 31.13 GiB used: 4.83 GiB (15.5%) Init: systemd
  target: graphical (5) Shell: Bash inxi: 3.3.26
```

Device states
```
   Usage     Device name
   4.6%        CPU misc
   4.6%        CPU core
   4.6%        DRAM
 100.0%        PCI Device: Shenzhen Longsys Electronics Co., Ltd. Lexar NM790 NVME SSD (DRAM
 100.0%        PCI Device: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630]
 100.0%        PCI Device: Intel Corporation Ethernet Connection (7) I219-V
 100.0%        PCI Device: Intel Corporation Cannon Lake PCH PCI Express Root Port #9
 100.0%        PCI Device: Intel Corporation Cannon Lake PCH Thermal Controller
 100.0%        PCI Device: Intel Corporation 8th Gen Core 4-core Desktop Processor Host Brid
 100.0%        PCI Device: Intel Corporation Cannon Lake PCH SPI Controller
 100.0%        PCI Device: Intel Corporation Cannon Lake PCH SATA AHCI Controller
 ```

Idle states
```
           Pkg(HW)  |            Core(HW) |            CPU(OS) 0
                    |                     | C0 active   3.4%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.1%    0.1 ms
C2 (pc2)   24.1%    |                     |
C3 (pc3)    1.1%    | C3 (cc3)    0.3%    | C3          0.7%    0.1 ms
C6 (pc6)   24.3%    | C6 (cc6)   27.9%    | C6         34.2%    0.6 ms
C7 (pc7)    0.0%    | C7 (cc7)   56.8%    | C7s         0.0%    0.0 ms
C8 (pc8)   27.6%    |                     | C8         38.7%    1.1 ms
C9 (pc9)    0.0%    |                     | C9          0.5%    2.7 ms
C10 (pc10)  0.0%    |                     |
                    |                     | C10        21.9%    9.6 ms
                    |                     | C1E         0.7%    0.1 ms

                    |            Core(HW) |            CPU(OS) 1
                    |                     | C0 active   1.7%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.0%    0.1 ms
                    |                     |
                    | C3 (cc3)    0.1%    | C3          0.3%    0.1 ms
                    | C6 (cc6)   14.0%    | C6         16.8%    0.6 ms
                    | C7 (cc7)   78.9%    | C7s         0.0%    0.0 ms
                    |                     | C8         15.6%    1.0 ms
                    |                     | C9          0.0%    0.0 ms
                    |                     |
                    |                     | C10        65.2%   14.0 ms
                    |                     | C1E         0.5%    0.1 ms

                    |            Core(HW) |            CPU(OS) 2
                    |                     | C0 active   1.9%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.3%    0.3 ms
                    |                     |
                    | C3 (cc3)    0.1%    | C3          0.1%    0.1 ms
                    | C6 (cc6)    5.9%    | C6          7.0%    0.6 ms
                    | C7 (cc7)   89.5%    | C7s         0.1%    1.4 ms
                    |                     | C8          7.9%    1.0 ms
                    |                     | C9          0.1%    1.4 ms
                    |                     |
                    |                     | C10        82.5%   20.8 ms
                    |                     | C1E         0.3%    0.1 ms

                    |            Core(HW) |            CPU(OS) 3
                    |                     | C0 active   3.3%
                    |                     | POLL        0.0%    0.0 ms
                    |                     | C1          0.1%    0.1 ms
                    |                     |
                    | C3 (cc3)    0.2%    | C3          0.3%    0.1 ms
                    | C6 (cc6)   17.9%    | C6         21.7%    0.6 ms
                    | C7 (cc7)   71.4%    | C7s         0.1%    1.4 ms
                    |                     | C8         25.5%    1.0 ms
                    |                     | C9          0.7%    2.8 ms
                    |                     |
                    |                     | C10        48.0%   13.6 ms
                    |                     | C1E         0.4%    0.1 ms
```

ValeZAA, 2025-04-12:
Last version of powertop is 2.15... Why are using ancient 2.14? Also last kernel is 6.13.10 in debian experimental. Install both of those from 

apt-get install -t experimental powertop

ValeZAA, 2025-04-13:
Not a bug: Z390 is both Cannon lake and coffee lake:

https://www.reddit.com/r/intel/comments/6tonb0/comment/dlm9p3c/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

> I have an Intel Core i3-8100 Coffee Lake CPU. Powertop, however, now shows "Cannon Lake"

No, it does not show anything, and iGPU it sees correctly:
PCI Device: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630

fenrus75, 2025-04-13:
the GPU descriptions come from the pci.ids file from your linux
distribution -- not directly from powertop.
However they are "cosmetic" in the sense that there is no functional
meaning inside powertop to what exactly is listed..

ValeZAA, 2025-04-13:
You did not get it. There is actually no any CPU shown on the logs, so bug report makes no sense. Only PCH and iGPU. And it is correct for PCH.

Leitner1, 2025-04-16:
Sorry for late replay. 
I was still running Powertop 2.14 because it was working fine and is included in the Debian package lists. 
I have now compiled 2.15 from this git repository and updated Linux to the latest available kernel 6.8.12-9-pve. But the result is the same. I can't get deeper than HW package state C8.

What I have been determine is, that since I can no longer get into HW package state C10, it shows “Intel Corporation Cannon Lake” everywhere where it was “Coffee Lake”. 
Also, when I run `powertop --auto-tune`, I get the following message:
```
modprobe cpufreq_stats failed
Loaded 0 prior measurements
Cannot load from file /var/cache/powertop/saved_parameters.powertop
File will be loaded after taking minimum number of measurement(s) with battery only 
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask f
RAPL device for cpu 0
RAPL Using PowerCap Sysfs : Domain Mask f
Devfreq not enabled
glob returned GLOB_ABORTED
Cannot load from file /var/cache/powertop/saved_parameters.powertop
File will be loaded after taking minimum number of measurement(s) with battery only 
Leaving PowerTOP
```

Here the complete HTML report from powertop: https://www.dropbox.com/scl/fi/20qbw17etnwnaal0ibazl/powerreport20250416.html?rlkey=zkn1vss8d594b42m2jvfvdf08&st=jj2n42os&dl=0

fenrus75, 2025-04-16:
the reporting comes from the pci.ids file (in ./usr/share/hwdata) not from
powertop or the kernel

but the actual name of things is not important for powertop itself, nor for
the kernel's power management algorithms -- it's a red herring to whatever
problem you have outside the cosmetics of the name on the screen

Leitner1, 2025-04-18:
OK, thanks for the explanation! Any idea how I can get the system back to Pkg HW C10?