fenrus75/powertop #179: Unusually high battery discharge rate reported by powertop

sidtronics, 2025-06-02:
Device: HP Probook 440 G6
OS: Arch LInux

[image]

I have tried calibrating using `powertop --calibrate` but it changes nothing.

sidtronics, 2025-07-04:
Apologies for the late reply.
```
$ uname -a

Linux Pb440 6.15.4-arch2-1 #1 SMP PREEMPT_DYNAMIC Fri, 27 Jun 2025 16:35:07 +0000 x86_64 GNU/Linux
```

```
$ upower -e

/org/freedesktop/UPower/devices/battery_BAT0
/org/freedesktop/UPower/devices/line_power_AC
/org/freedesktop/UPower/devices/DisplayDevice
```

```
$ upower -i /org/freedesktop/UPower/devices/battery_BAT0

native-path:          BAT0
  vendor:               Hewlett-Packard
  model:                Primary
  serial:               01396 2019/09/15
  power supply:         yes
  updated:              Fri 04 Jul 2025 02:19:59 PM IST (15 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    energy:              20.7092 Wh
    energy-empty:        0 Wh
    energy-full:         35.0081 Wh
    energy-full-design:  45.045 Wh
    energy-rate:         6.92692 W
    voltage:             11.407 V
    charge-cycles:       626
    time to empty:       3.0 hours
    percentage:          59%
    capacity:            77.7179%
    technology:          lithium-ion
    icon-name:          'battery-good-symbolic'
  History (rate):
    1751618999	6.927	discharging
    1751618969	2.768	discharging
    1751618939	8.305	discharging
    1751618909	2.955	discharging
    1751618904	3.592	discharging
```

paulmenzel, 2025-07-04:
Sorry, of course, also what PowerTOP version do you use?

The output comes from `src/process/do_process.cpp`:

https://github.com/fenrus75/powertop/blob/49045c0c8ca7d3f47b8e289a9436d5ab2f4e93d9/src/process/do_process.cpp#L823-L840

`global_power()` is implemented in `src/measurement/measurement.cpp`:

https://github.com/fenrus75/powertop/blob/49045c0c8ca7d3f47b8e289a9436d5ab2f4e93d9/src/measurement/measurement.cpp#L77-L96

Maybe you can use a debugger like GNU gdb to add a break point there, and try to figure out, what measurement causes the high value? (Sorry for not being able to give better support.)

sidtronics, 2025-07-04:
Thanks for guiding me. I followed your steps and here's what I found:

The rate is calculated by sysfs_power_meter class:
[image]

It is calculated using either "power_now" or "current_now" sysfs attributes:
https://github.com/fenrus75/powertop/blob/master/src/measurement/sysfs.cpp#L71-L99

So I tried to inspect the value of current_now (I don't have power_now attribute) however I got:
```
$ cat /sys/class/power_supply/BAT0/current_now
cat: /sys/class/power_supply/BAT0/current_now: No such device
```

Somehow it reports correctly when running on AC:
```
$ cat /sys/class/power_supply/BAT0/current_now
376000
```

I researched further about this behavior and found this bug report which is very similar to my case:
https://bugzilla.kernel.org/show_bug.cgi?id=83411

So maybe this is kernel or BIOS issue? But either way I think powertop is not able handle "No such device" error and parsing garbage values in rate instead.